mariolukas/FabScanPi-Server #5: server process high CPU usage with no active connections

dlarue, 2016-01-27:
I noticed a high CPU usage with a client open but no action happening(home screen). Even after I shut down the client window there was no change in the CPU utilization of the fabscanpi-server process. The CPU chip was warm to the touch.

theouf, 2016-03-11:
Hello,
I've got the same issue. I use the fabscanpi_basic_jessie_2016-02-27-01-12.img image. Do you find why ? it's strange to use more than 90% of CPU in waiting state.

mariolukas, 2016-03-11:
Not yet. But i have an idea and i am working on that.

mariolukas, 2016-03-11:
Fixed some threading things, and found out that this bug is related to #2 . Finally i found a workaround for #2 and #5 by adding an directive for mjpeg, which opens mjepeg stream in an iframe. When the settings dialog is closed, the iframe is destroyed and the mjpeg stream is forced to close. It is very ugly but it works, until mjpeg bugs in several browsers are fixed. 
I will give it some more tests and add it to the new release.

closed 2016-03-12 by mariolukas