mariolukas/FabScanPi-Server #7: change configuration file in UI

mariolukas, 2016-02-06:
A feature/dialog to change the fabscanpi-settings within the UI would be great.

johnelle, 2016-02-07:
Well just to be clear all I was asking about was the versioning.  If the UI could display the various version numbers it would be nice.  Currently the UI displays a version number on the tab via html.  The server has a version number associated with the package and I am guessing there is a version associated with the firmware(?) 

Might be simpler to just increment the version number on everything in sync even if one of the pieces doesn't change they're all 0.6 'cause they were tested as a single unit (for example)

mariolukas, 2016-02-07:
Version numbering:
I understood. I think that makes sense. 

Anyway i created this issue, because i think it is easier for people who are less linux skilled to change the config file settings within the UI. 

By the way thank you for your suggestions. That helps a lot!

mariolukas, 2016-07-28:
Helpful javascript libraries:
https://github.com/josdejong/jsoneditor
https://github.com/jdorn/json-editor