mariolukas/FabScanPi-Server #8: Enhancement Request:  Wireless Subsystem Added to OS

johnelle, 2016-02-07:
Tried to get my system running on wireless and having no luck.  Installed wireless-tools.  Created config files.  Still not able to get it working with two dongles.  Usually very easy process.

Seems to be some things missing from the Raspian image (drivers?) that don't allow this.

mariolukas, 2016-02-07:
Which wifi Dongle?

johnelle, 2016-02-08:
WiPi  and Edimax

I just retried.  Setup the wipi on an older pi using the supplicant
method.  Copied it exactly on the fabscan and it didn't come up.  Tried the
alternative not using supplicant (everything in interfaces) still no go.

No indicator light on the dongle.  No listing of wlan on the ifconfig.  No
errors from ifup / ifdown.

mariolukas, 2016-02-08:
I have an Edimax wifi dongle here, i will give it a try.

mariolukas, 2016-02-11:
I have tried some things, worked for me. First of all try to install

sudo apt-get install wpasupplicant

Then add the following content at the end of your /etc/network/interfaces

auto wlan0
allow-hotplug wlan0
iface wlan0 inet dhcp
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
iface default int dhcp

Restart your network with

sudo /etc/init.d/networking restart

Next open /etc/wpa_supplicant/wpa_supplicant.conf if it does not exists create the file and add the following content.

update_config=1
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
eapol_version=1

network={
ssid="YOUR_SSID"
psk=YOUR_SECRET
proto=RSN
key_mgmt=WPA-PSK
pairwise=CCMP
auth_alg=OPEN
}

After you added these lines, you can try to get your wifi dongle up.

sudo ifup wlan0

If it does not work out of the box, a reboot did the job for me.

Optional: You can deactivate the wifi dongle sleep mode by adding a line to /etc/modprobe.d/8192cu.conf
This file didn't exist in my image, so i created it. Add the following lines to it and reboot the system or reload the kernel modules.

options 8192cu rtw_power_mgnt=0 rtw_enusbss=0

This is what worked for me with an EDIMAX dongle. If you can confirm that i will add some config templates to the next sd-card image.

Sorry i don't know how deep your debian/linux skills are. So i decided to writ all the stuff more detailed, maybe other users want to know all the details.

mariolukas, 2017-07-11:
This will be fixed with #102

closed 2017-07-11 by mariolukas