mariolukas/FabScanPi-Server #9: Enhancement Request:  Wireless Subsystem Added to OS

johnelle, 2016-02-07:
Tried to get my system running on wireless and having no luck.  Installed wireless-tools.  Created config files.  Still not able to get it working with two dongles.  Usually very easy process.

Seems to be some things missing from the Raspian image (drivers?) that don't allow this.

mariolukas, 2016-02-07:
Solved? Let me know how... Maybe i can add the needed packages by default for other people with the same problem.

johnelle, 2016-02-08:
Not solved.  I added wireless-tools but that just helped me with the
configuration.  Not sure what else is missing.  Clearly some drivers missing

mariolukas, 2016-02-11:
I have tried some things, worked for me. First of all try to install 

sudo apt-get install wpasupplicant

Then add the following content  at the end of your /etc/network/interfaces

auto wlan0
allow-hotplug wlan0
iface wlan0 inet dhcp
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
iface default int dhcp

Restart your network with 

sudo /etc/init.d/networking restart

Next open /etc/wpa_supplicant/wpa_supplicant.conf if it does not exists create the file and add the following content.

update_config=1
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
eapol_version=1

network={
ssid="YOUR_SSID"
psk=YOUR_SECRET
proto=RSN
key_mgmt=WPA-PSK
pairwise=CCMP
auth_alg=OPEN
}

After you added these lines, you can try to get your wifi dongle up.

sudo ifup wlan0

If it does not work out of the box, a reboot did the job for me. 

Optional: You can deactivate the wifi dongle sleep mode by adding a line to /etc/modprobe.d/8192cu.conf
This file didn't exist in my image, so i created it. Add the following lines to it and reboot the system or reload the kernel modules.

options 8192cu rtw_power_mgnt=0 rtw_enusbss=0

This is what worked for me with an EDIMAX dongle. If you can confirm that i will add some config templates  to the next sd-card image.

Sorry i don't know how deep your debian/linux skills are. So i decided to writ all the stuff more detailed, maybe other users want to know all the details.

closed 2016-02-11 by mariolukas

johnelle, 2016-02-12:
Unfortunately I have not been successful with this.  Still missing
something.

I re-imaged the card and did updates (just picked up a new one today in
fact).  Set-up the WPA-supplicant stuff.  For the Edimax dongle I can get
it configured so I can list the access points but it won't connect to any
of them for some reason.

The WiPi dongle won't even configure (i.e. never shows up in ifconfig)

Oh well.