mariolukas/FabScanPi-Server #11: camera preview window freezes on start of new scan

dlarue, 2016-02-14:
after a completed scan, when another scan is attempted the preview camera window does not update. Using Chrome(Version 48.0.2564.109 (64-bit)) on Ubuntu 14.04 LTS.

A web page reload will cause a working camera preview window

mariolukas, 2016-02-14:
This is similar to #2. It is on the top of the todo list now.

dlarue, 2016-02-14:
If that one happens or is triggered after a scan and upon the start of a new scan then I agree.

mariolukas, 2016-02-14:
It is not the same, but it is the same part of the software which causes this error. I am using mjpeg for streaming. The problem is caused in opening and closing mjpeg streams in different browsers. Some keep the stream open, even if the streaming window is not active.

dlarue, 2016-02-14:
When I get a chance I will test it in Firefox. I'm currently using Chrome.

dlarue, 2016-02-22:
it works again. I both build and tried a new image and reseated the camera ribbon cable on the rPi. I don't know what solved the problem but this one can be closed.

dlarue, 2016-02-22:
oops, wrong issue closed.  I was thinking this was #13, no camera preview image. I didn't get a chance to try Firefox yet.

mariolukas, 2017-07-11:
Fixed

closed 2017-07-11 by mariolukas