mariolukas/FabScanPi-Server #12: Internet connection required for operation

dlarue, 2016-02-14:
the client browser window requires an internet connection to operate.

I've setup the Fabscan Pi system on a wireless access point(WAP) without an Internet connection( isolated LAN ). In this configuration the browser windows to the Fabscan Pi server will not load the lower control area so scanning is not possible.  As soon as a WAN connection is available to the LAN devices(client) the lower control area will populate on a page reload.

mariolukas, 2016-02-14:
font-awesome.min.css contains all the icon stuff. It is load from a CDN and not statically in the package. I think adding the css to the package will fix it.

mariolukas, 2016-02-14:
I fixed it in the current master branch. I added font awesome lib to the fabscan fronted libraries, there is an additional font folder in the fronted now. The css which was linked in index.html is compiled to lib.css now.  You can test it by copying the git src/www folder to your /usr/local/fabscanpi/www/ folder. If it works it will be part of the next package release. Thanks for reporting.

dlarue, 2016-02-17:
I did do the copy and it did appear to work, ie without an Internet connection the bottom control area populated perfectly. There were other issues which prevent more testing but so far looking good on this issue.

closed 2016-02-17 by mariolukas