mariolukas/FabScanPi-Server #13: Camera preview window shows no images

dlarue, 2016-02-17:
a recent change seems to have effected the preview section of either the client or the server( I know, duh :) but for what ever reason the video preview window does not show valid video.

mariolukas, 2016-02-17:
Can you wait 2 or 3 seconds after opening the settings dialog. Sometimes it will need some time to initialize the preview. In my setup with a fresh installation of the latest image it works. 

If that is the problem i will add a spinner or initializing message to the black window to inform the user about what's going on...

dlarue, 2016-02-17:
I had the window open for a few minutes so it's not a delay. But I did see someone mention the cable coming loose and I have been swapping uSD cards lots. I will check this tonight since I'll be at the lab for a 3D printer/prosthetics meetup.

mariolukas, 2016-02-17:
If the camera cable is coming loose you should see an entry in the log, right after starting the server it should throw an camera device error. 

Stange is that Willie decsribed a similar error. My setup is working quite nice.

dlarue, 2016-02-17:
I saw no log info on the camera but will check again tonight.

mariolukas, 2016-02-17:
You can use raspstill to be sure that your cam works.

dlarue, 2016-02-17:
As you know, I don't have and Internet connection on that machine so either I have to add it to my build script and/or wait until it's in a standard image. I'll hack my build script further and add the raspistill package.

mariolukas, 2016-02-17:
I added a new script to delivery folder. 019_tools.sh 
This script contains raspstill. I also added mc, htop and minicom.

dlarue, 2016-02-18:
fantastic, thanks.  It would be helpful if it is the last thing to load too. This way we can add awk/sed scripts to it to do things like update the Serial device.

mariolukas, 2016-02-22:
Obviously solved?

dlarue, 2016-02-22:
Yes, I believe so. I reseated the camera cable on the rPi and tried a different build image and while I could not get raspistill installed, when I tested FabScan the preview was there.

closed 2016-02-22 by dlarue

mariolukas, 2016-02-22:
sounds good