mariolukas/FabScanPi-Server #15: Cleanup config file.

mariolukas, 2016-02-18:
Default parameters for custom setups should not appear in config file. 
Parameters like serial, camera, process_number can be removed, 
fabscanpi-server should use default values if they are not set.

dlarue, 2017-05-22:
But shouldn't defaults be set so it acts as a form of documentation of available options?

Otherwise, the source code would have to be probed to find out what options are available as I had to do when I noticed flash_baudrate was handled and defaulted to 115200.  Often what's done is the option is commented out in the configuration file and defaults taken.

The next option would be keeping the webpages/wiki/etc updated with these options specifically for use of older system or custom configurations.

I have not checked to see if the JSON parser used can handle comments in the file. So far I've seen that the JSON writer doesn't do formatting as seen in my default.settings.json file:

pi@fabscanpi:/etc/fabscanpi$ cat default.settings.json
{"led": {"blue": 152, "green": 149, "red": 152}, "center": 0.5, "color": false, "line_coords_buffer": 5, "laser_positions": 1, "backwall": {"laser": {"y": 12.937499999999998, "x": -3.935185185185185, "z": 0.0}, "margin": 15, "laser_pixel_position": 444.0, "laser_angle": 35.005691716773715, "offset": 100}, "crop": {"upper": -128, "lower": -810}, "camera": {"saturation": 0, "contrast": 0, "brightness": 50}, "file": "/etc/fabscanpi/default.settings.json", "threshold": 106, "resolution": 12, "percent_top_pixels": 0.1}

which originally looked like this:
pi@fabscanpi:/etc/fabscanpi$ cat default.settings.jsonORG 
{
     "led":{
       "red": 0,
       "green": 0,
       "blue": 0
     },
     "line_coords_buffer": 5,
     "percent_top_pixels": 0.1,
     "center": 0.50,
     "threshold": 100,
     "crop": {
       "upper": -128,
       "lower": -810
     },
     "backwall": {
       "laser_pixel_position" : 500,
       "laser": {
         "x": 0.0,
         "y": 11.0,
         "z": 0.0
       },
       "laser_angle": 32.506192762678744,
       "margin": 15,
       "offset": 100
     },
     "camera":{
       "brightness" : 50,
       "contrast": 0,
       "saturation": 0
     },
     "resolution": 8,
     "laser_positions": 1,
     "color" : "True"
}

mariolukas, 2017-07-11:
Configs are written in a readable format now. Most of the old measurement related params are not needed anymore.

closed 2017-07-11 by mariolukas