mariolukas/FabScanPi-Server #38: Add documentation PI HAT Connection

mariolukas, 2016-07-05:
- power jack polarity
- wireing diagram 
- laser secure switch (bridge)
- jst connector polartity (howto connect led stripe/ring)
- silent step stick driver configuration/assembly

jenshackel, 2016-09-11:
This will be my next "construction site".

Cysign, 2016-11-28:
- Add info about voltage/maximum current for the laser connectors

- Add instructions to setup the stepper current properly

- Add checklist 'before you turn on your machine'

Cysign, 2016-11-28:
- Add information if a resistor for an DIY-WS2812 strip is needed on the signal

- Does the Hat deliver power to the Pi?

- Add: Use some tape on audiojack & hdmi port to properly isolate hat from Pi !

mariolukas, 2016-11-28:
Thank you for your suggestions, we will add those points to the hardware related docs.