mariolukas/FabScanPi-Server #40: Connection Problem with Arduino

mariolukas, 2016-07-13:
avrdude reports sync problems with the arduino. Try other baudrates. 
If higher baudreates work, add optional configuration of flash baudrates
to config file.

brummbaer, 2016-09-07:
Any news on this one? I can not get it to work even with 57600 baud or 115200 baud. I installed your latest image.
My settings are: 
```
"serial" : {
     "baudrate" : 57600,
     "port": "/dev/ttyACM0",
     "autoflash": "False"
   }
```
with the settings above i can turn the laser on with:
`echo "M21" > /dev/ttyACM0`
but still got the error:
```
2016-09-07 08:13:40,848 - fabscan.controller.FSSerial - DEBUG - Latest available firmware version is: v.20160620
2016-09-07 08:13:42,419 - fabscan.controller.FSSerial - ERROR - No Arduino compatible device found on port /dev/ttyACM0
2016-09-07 08:13:42,421 - fabscan.controller.FSSerial - ERROR - Fatal Arduino connection error....
```
and top shows autoreset is on 100% cpu after boot for about a minute.                                                                                             

How do you test whether it is arduino compatible or not? Maybe i can find something.
I changed some setting in the arduino files.
in FabScanArduinoFirmware:
`Serial.begin(57600);`
and ports in configuration.h:
```
#define LIGHT_PIN A3 // old A1
#define RIGHT_LASER_PIN 12 // old A4
#define LEFT_LASER_PIN 9 // old 9
...
//Stepper 1 as labeled on Shield, Turntable
#define ENABLE_PIN_0  8 //old 2
#define STEP_PIN_0    2 //old 3
#define DIR_PIN_0     5 //old 4

```

mariolukas, 2016-09-07:
Are you using an Arduino or the FabScanPi HAT?

brummbaer, 2016-09-07:
An Arduino with motor shield.

brummbaer, 2016-09-08:
With the latest FabScanPi-Server_0.3.0_all from today it detects Arduino. Laser/Stepper/Lights are working.

mariolukas, 2016-09-08:
All things are working related to this issue?

brummbaer, 2016-09-08:
I think so. 115200 baud and it works flawlessly.
```
2016-09-08 07:58:02,431 - fabscan.scanner.laserscanner.FSSerial - DEBUG - Latest available firmware version is: v.20160620
2016-09-08 07:58:02,729 - fabscan.scanner.laserscanner.FSCamera - INFO - Camera module ready...
2016-09-08 07:58:03,618 - fabscan.scanner.laserscanner.FSCamera - DEBUG - PI Camera Moule ready.
2016-09-08 07:58:04,137 - fabscan.scanner.laserscanner.FSSerial - DEBUG - Trying to connect Arduino on port: /dev/ttyACM0
2016-09-08 07:58:06,147 - fabscan.scanner.laserscanner.FSSerial - DEBUG - Installed firmware version: v.20160620
2016-09-08 07:58:06,149 - fabscan.scanner.laserscanner.FSSerial - INFO - FabScanPi is connected to Arduino or FabScanPi HAT on port: /dev/ttyACM0
2016-09-08 07:58:08,272 - fabscan.scanner.laserscanner.FSHardwareController - DEBUG - Hardware controller initialized...
2016-09-08 07:58:08,279 - fabscan.scanner.laserscanner.FSScanProcessor - INFO - Laser Scan Processor initialized..
2016-09-08 07:58:08,289 - fabscan.FSScanner - INFO - Scanner initialized...
2016-09-08 07:58:08,290 - fabscan.FSScanner - INFO - Number of cpu cores: 4
```

```
pi@raspberrypi:~ $ stty -F /dev/ttyACM0
speed 115200 baud; line = 0;
min = 0; time = 0;
-brkint -icrnl -imaxbel
-opost
-isig -icanon -iexten -echo -echoe -echok -echoctl -echoke
```

```
pi@raspberrypi:~ $ sudo avrdude -p m328p -b 115200 -carduino -P /dev/ttyACM0

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f

avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.
```

mariolukas, 2016-09-08:
By the way, the new version uses 115200 Baud by default now. 
I added a new config setting, which is optional and default value is 115200.
   
"serial" : {
     "baudrate" : 115200,
     "autoflash": "True",
     "port": "/dev/ttyAMA0",
     "flash_baudrate": 115200    <------ this one is new, by default not shown and 115200
}

closed 2016-09-08 by mariolukas