mariolukas/FabScanPi-Server #71: Manual / Instructions / Issues FabScan Pi hat

Cysign, 2016-11-28:
Hi, I just assembled my FabScan Pi and after a lot of research I wonder if my laser(s) will worp properly.

As the scematics (https://github.com/watterott/RPi-FabScan-HAT/blob/master/hardware/RPi-FabScan-HAT_v10.pdf) shot an 100Ohm resistor in front of the laser any my 4V/45mA Laser only requires an 22Ohm resistor to work properly, I wonder what to do.

There's no detailed informaton about choosing the laser for the project. As I've been playing with David3D Laserscanner in 2010 I still have had those lasers.

Another riddle is how to adjust the power of the A4988 with the FabScan Pi hat.
Using my own circuit I can measure the current, but as I didn't start the FabScan Pi yet, I want to be well prepared.

I know github isn't the right place for those questions...but I cannot find and forum or detailes information on this.

Placement of the power socken on the fabscan pi hat is incredibly strange. I cannot connect any powersupply because it's too close to the stepper for the rotary table.

I had to bend my wire end sleeves on the laser-terminals, and the stepper terminals.
Ths design needs to be updated!

mariolukas, 2016-11-28:
Hey, 
thank you for reporting all those things. I will update the docs as soon as possible. Please send me an E-Mail to info@fabscan.org ( i will send you an invitation to our slack channel )

By the way: GitHub is a quite good place to report your issues :)

closed 2019-02-18 by mariolukas

Cysign, 2019-02-18:
Already there - waiting for the new release ;)