mariolukas/FabScanPi-Server #87: No Connection to Arduino when there is - Sanguino

dlarue, 2017-05-09:
I have the latest firmware on the Sanguino and installed FabScanPi from git and my default.config.json file in /etc/fabscanpi "autoflash" is set to False.  The server web front end says the Arduino/Sanguino is not connected yet I can operate the system via minicom as also shown below.

BTW, I see a process trying to run avrdude and if this is what validates a connection it won't work with devices other than a 328 arduino..

It runs this and fails:
pi@raspberrypi:~ $ avrdude -p m328p -b 115200 -carduino -P/dev/ttyUSB0
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding

while I can run this and pass:
pi@raspberrypi:~ $ avrdude -p m1284p -b 57600 -carduino -P/dev/ttyUSB0
avrdude: AVR device initialized and ready to accept instructions
Reading | ################################################## | 100% 0.04s
avrdude: Device signature = 0x1e9705
avrdude: safemode: Fuses OK (E:00, H:00, L:00)
avrdude done.  Thank you.

ps -fe showing avrdude run:
-----------------------------------------------------------------------------------
root      1971     1 75 11:07 ?        00:00:02 /usr/bin/python /usr/bin/fabscanpi-server --config=/etc/fabscanpi/default.config.json --
root      1988  1971  5 11:07 ?        00:00:00 avrdude -p m328p -b 115200 -carduino -P/dev/ttyUSB0
root      1991     1  1 11:07 ?        00:00:00 /sbin/agetty --keep-baud 115200 38400 9600 ttyAMA0 vt102
pi        1992  1024  0 11:07 pts/0    00:00:00 ps -fe
pi@raspberrypi:~ $ 
-------------------------------------------------------------------------------------

minicom screen capture verifying the arduino is operating the G-Code correctly( note spelling error ):
-----------------------------------------------------------------------------------------
M100
    FabScan G-CODDE Interpreter Version: v.20170326
Commands:
G00 [T(steps)]; - linear move
G01 [T(steps)] [L(steps)] [F(feedrate)]; - move                                  
M17; - enable motors                                                             
M18; - disable motors                                                            
M19; - turn left laser on                                                        
M20; - turn left laser off
M21; - turn right laser on
M22; - turn right laser off
M100; - this help message
M114; - report position and feedrate
>
G00 T5
      >
G01 T100
        >
G01 T-100
         >
M18
   >
-----------------------------------------------------------------------------------------------------

copy of my config file:
----------------------------------------------------------------------------------------------------
pi@raspberrypi:~ $ cat /etc/fabscanpi/default.config.json
{
   "folders" : {
    "www": "/var/www/",
    "scans": "/var/scans/"
   },
   "serial" : {
     "baudrate" : 115200,
     "autoflash": "False",
     "port": "/dev/ttyUSB0"
   },
   "camera" : {
     "type" : "PICAM",
     "rotation_angle" : 90,
     "preview_resolution":{
        "width": 240,
        "height": 320
     },
     "resolution":{
          "width": 972,
          "height":1296
      },
      "position":{
          "x": 0.0,
          "y": 6.6,
          "z": 24.6
      },
      "frame":{
          "dimension": 27
      }
   },
   "laser": {
      "position":{
        "x": 8.3,
        "y": 5.3,
        "z": 20
      },
      "angle": 28.7,
      "detection_limit": 0.20,
      "rotation_steps": 5,
      "steps": 3200
   },
   "turntable":{
     "position": {
       "x": 0.0,
       "y": 0.0,
       "z": 7.5
     },
     "steps":3200
   },
   "scanner": {
      "origin":{
        "y" : 0.69
      }
   },
  "process_numbers": 4,
   "meshlab":{
     "path": "/usr/bin/"
   }
}
---------------------------------------------------------------------------------------------------

dlarue, 2017-05-12:
What I found was the data is returned in 3 lines but the code in FSSerial is only making one call to readline(). 

The data coming back from an M200\n or M200;\n looks like this using the firmware version in the returned data:
Line: ['M', '2', '0', '0', '\n']
Line: ['v', '.', '2', '0', '1', '7', '0', '3', '2', '6', '\r', '\n']
Line: ['>', '\r', '\n']

I don't know how this would work on the FabScanPi Hat Arduino but not with the Sanguino using the same firmware version.

https://github.com/mariolukas/FabScanPi-Server/blob/autocalibration/src/fabscan/scanner/laserscanner/FSSerial.py#L140

dlarue, 2017-05-15:
I've gone back and simplified the task of getting a connection and getting the version from an M200 request. Basically using a loop after the initial "/r/n/r/n" wakeup sequence. So besides patching FSSerial.py to get the flash_baudrate and devicemodel from the default.config.json file, it also has checkVersion() which loops like so:
`    def checkVersion(self):
        if self._serial:
            self.send("\r\n\r\n")
            time.sleep(.1) # Wait for FabScan to initialize

            tryCt = 0
            while tryCt < 10:
                self.send("M200;\n")
                time.sleep(.1) # Wait for FabScan to initialize
                value = self._serial.readline()
                value = self._serial.readline()
                value = value.strip()
                #self._logger.info("Hello read M200-value:"+str(value))
                if value != "":  # we got something, break out of loop
                    break #tryCt = 10
                else:
                    # try another time
                    self._serial.flushInput() # Flush startup text in serial input
                    tryCt += 1

            if value != "":
                return value
            else:
                return "None"
        else:
            return "None"
`

mariolukas, 2017-05-16:
Thanky Doug for all those work! I am stuggeling with a similar problem. Inconsistent connections to /dev/ttyAMA0. During my "weekend research" i found out that my problems are related to Raspberry Pi3's BT device which uses the same UART. I am getting problems even when i deactivate the BT device in /boot/config.txt 

I will try to fix that and then test and merge your bug fix.

dlarue, 2017-05-17:
I can do a fork and a pull request on it so you can see the diffs of what I've changed. But I had used the latest prebuilt image so it's from master and not one of the more recent environments( like auto-calibration).

Hopefully there is a way to assign the BT device to another UART since you can't so easily reroute the FabScanPi Hat.

mariolukas, 2017-05-17:
There are some blogs, where different solutions are described. But nothing worked for me. But i will do another try with a fresh installtion this weekend.

mariolukas, 2017-05-18:
I think i found a solution last night. It is a combination of changes in Firmware, FSSerial.py and Raspbian Boot Settings. I will do some more tests today and if it works stable i will commit and push the changes.

dlarue, 2017-05-18:
Great!  I forked, updated and created a pull request.  It includes changes to the init startup script to pull data from the default.config.json file since it too does comms with the Arduino.  Also added using that config data for firmware and device specifics and adding the loop to FSSerial.py so it has a better chance of getting the M200/version number from the Arduino and Sanguino.

It's setup for default use, ie the FabScanPi Hat so should work for you out of the box but for others it's just a tweak of the default.config.json file.

Also, I ordered a rPi3 today so I might get a chance to verify your changes. I have been running off an rPi 2 B.

dlarue, 2017-05-24:
I tried out the autocalibration branch  and I'm not getting too far. I got it running using this start script run from FabScanPi-Server/src directory:
#!/bin/bash
/usr/bin/python ./fabscanpi-server --debug --loglevel info --host fabscanpi.local --port 8080 --camera PICAM --config /home/pi/Testing/etc/fabscanpi/default.config.json --settings /home/pi/Testing/etc/fabscanpi/default.settings.json --logfile /home/pi/fabscanpi.log --pid /home/pi/fabscanpi.pid

I'm using pretty much the default config and settings JSON files except in config:
I change the camera resolution to "width": 1296,  "height": 972
I changed the serial:         "autoflash": "False",  "port": "/dev/ttyUSB0"

But I don't even think I'm getting very far into the software since my log is short and has this:
cat ~/fabscanpi.log 
2017-05-24 04:09:44,362 - fabscan.server - INFO - FabScanPi-Server 0.3.2-alpha.1
2017-05-24 04:09:45,624 - fabscan.server.websockets - INFO - Websocket Server started on port 8010
2017-05-24 04:09:45,733 - fabscan - CRITICAL - Fatal error: max() arg is an empty sequence
2017-05-24 04:09:45,868 - fabscan.scanner.laserscanner.FSCamera - INFO - Camera module ready...

mariolukas, 2017-05-24:
This is my fault. The firmware folder is not in the branch.

dlarue, 2017-05-24:
ok, I'll update my firmware and try again.

dlarue, 2017-05-24:
It's running but with issues.  I don't see any camera preview and while the LED lighting turns on when the Calibration option is selected, they don't operate when the LED lighting option sliders are used.

The camera is a V1 rPiCam and PICAM is set in the config and on the commandline. I have the config changed to match the V1 PICAM resolution of 1296x972.

mariolukas, 2017-05-25:
Due this is a dev branch, it may be that there are issues. I added the firmware folder now. The latest firmware hex file is included now.  Is there a hourglass turning where the camera image should appear?
It may take a moment until the preview is available.

I never tested it with the pi cam 1 but it should work out of the box, cause the server is using the official python pi cam lib. 

By the way check your mails.

dlarue, 2017-05-25:
yes, there is an hour glass which I've left active for a minute or two but I can leave it running longer. It used to be fairly quick to load up.

dlarue, 2017-05-26:
I can't seem to email you because Google's email servers are being flagged as SPAM senders...
Twice I received this type of thing from email sent to you:

There was a temporary problem delivering your message to info@mariolukas.de. Gmail will retry for 46 more hours. You'll be notified if the delivery fails permanently.

I will send from an account I seldom check and hope this eventually gets cleared up.

mariolukas, 2019-02-18:
Solved in current release

closed 2019-02-18 by mariolukas