mariolukas/FabScanPi-Server #89: Calibration confusion

dlarue, 2017-05-22:
Lots of the calibration items are well documented in the Software section here:
https://mariolukas.github.io/FabScanPi-Server/software/

But what I've not found was how to use the Calibration item with the horizontal green line and vertical green.  I've seen old documents which say the horizontal green line should be lined up behind the Turn Table at the corner of the platform and back of box. Another looks like it's aligned at the center of the turn table based on what looks like the duck being scanned.

Also not yet found is what the "Start Calibration" item does and how it effects the default.settings.json file.  I currently don't have a stepper motor on my laser and having a tough time getting the system calibrated to produce a well formed model.

mariolukas, 2017-05-23:
"Start Calibration" does the laser line detection and calculates the laser angle. In older software versions this was done before each scan. But in some cases ( when the object is too big ), the laser can not be detected on the back wall. This was the reason why the laser line detection ( angle calculation ) was separated from the scanning process. 

All the calibration related stuff in the current version is deprecated. The new  version of FabScanPi-Server will bring auto calibration, where a chessboard is placed on the turntable. All measurements will be done automatically. You can find a early pre release in the auto calibration branch https://github.com/mariolukas/FabScanPi-Server/tree/autocalibration
But there is no calibration until now. I am doing some final tests and then i will do a release.

closed 2017-05-23 by mariolukas

dlarue, 2017-05-23:
Can I ask about the horizontal green line?  Is it to line up with the center of the turn table, the back of it or the back corner(corner made from the turn table platform and back wall)?

I will try the auto-calibration branch as soon as I can figure out how to either build the executable(fabscanpi-server) or turn from source.

mariolukas, 2017-05-23:
Sorry forgot to answer exactly this question :) 
It should line up with the back of the turntable.

I am working on a testing repository which will contain all the unstable builds. This will be more convenience for testing the latest unreleased features.

dlarue, 2017-05-23:
Ok, thanks. So I had it wrong in both my tests(center and back corner). figures. :-)

That'd be helpful but I also have no problem learning the ins and outs of python.