mariolukas/FabScanPi-Server #90: Camera orientation, size and the frame dimension setting

dlarue, 2017-05-22:
Lots of the calibration items are well documented in the Software section here including the frame dimension setting:
https://mariolukas.github.io/FabScanPi-Server/software/

What I found using the FabScan Cube chassis is that the camera ribbon extends down but the default in the default.config.json file is for a 90 deg rotation and W/H are listed as 240/320 and 972/1296. Here's the camera section from the original default.config.json and just changing the 90 deg orientation to 0 deg results in a narrow viewing window and my frame dimension can only be set to 19.7 CM.

ORIGINAL:

   "camera" : {
     "type" : "PICAM",
     "rotation_angle" : 90,
     "preview_resolution":{
        "width": 240,
        "height": 320
     },
     "resolution":{
          "width": 972,
          "height":1296
      },
      "position":{
          "x": 0.0,
          "y": 6.6,
          "z": 24.6
      },
      "frame":{
          "dimension": 27
      }
   },

Swapping the W/H values does provide a view of the inside of the chassis such that a frame dimension value of 25.0 is measured and much closer to the default of 27 CM.  But, with these changes my preview window extends into the area of the controls. selecting Color checkbox is not possible.  See pic below:
[image]

dlarue, 2017-05-22:
From this document, I see that the 240x320 resolution(used in preview) is not supported by either of the rPi cameras(v1 or v2).  https://picamera.readthedocs.io/en/release-1.12/fov.html

I also saw where the original rPi firmware for the camera was incorrectly rotating the image 90 degrees but has supposedly been fixed.

mariolukas, 2017-05-23:
The Preview resolution is calculated by software.

mariolukas, 2017-05-23:
By the way the Screenshot does not look good. Which Chrome version are you using?

dlarue, 2017-05-23:
Version 58.0.3029.96 (64-bit)

dlarue, 2017-05-23:
Oh my, I just loaded up the front end in Firefox and it's much different. The preview window is still running over into the controls but the display shows the scan as it occurs.

closed 2019-02-18 by mariolukas