mariolukas/FabScanPi-Server #115: Scanning stops at Position 51

brummbaer, 2017-08-23:
I tried to calibrate the scanner but it stops at Postion 51.
Here is the log, nothing happens after this step for an hour.

```
2017-08-23 16:29:41,700 - fabscan.scanner.laserscanner.FSCalibration - DEBUG - Calibration Position 48 of 74
2017-08-23 16:29:43,307 - fabscan.scanner.laserscanner.FSCalibration - DEBUG - Pose detected...  
2017-08-23 16:29:43,813 - fabscan.scanner.laserscanner.FSSerial - DEBUG - M05 R50 G50 B50;
2017-08-23 16:29:45,639 - fabscan.scanner.laserscanner.FSSerial - DEBUG - G04 T-44.4444444444 F900;
2017-08-23 16:29:45,640 - fabscan.scanner.laserscanner.FSCalibration - DEBUG - Calibration Position 49 of 74
2017-08-23 16:29:47,366 - fabscan.scanner.laserscanner.FSCalibration - DEBUG - Pose detected...  
2017-08-23 16:29:47,872 - fabscan.scanner.laserscanner.FSSerial - DEBUG - M05 R50 G50 B50;
2017-08-23 16:29:49,696 - fabscan.scanner.laserscanner.FSSerial - DEBUG - G04 T-44.4444444444 F900;
2017-08-23 16:29:49,697 - fabscan.scanner.laserscanner.FSCalibration - DEBUG - Calibration Position 50 of 74
2017-08-23 16:29:53,886 - fabscan.scanner.laserscanner.FSCalibration - DEBUG - Pose detected...  
2017-08-23 16:29:55,238 - fabscan.scanner.laserscanner.FSSerial - DEBUG - M05 R0 G0 B0;
2017-08-23 16:29:55,240 - fabscan.scanner.laserscanner.FSCalibration - DEBUG - Starting laser capture...
2017-08-23 16:29:55,744 - fabscan.scanner.laserscanner.FSSerial - DEBUG - M21;
2017-08-23 16:30:00,775 - fabscan.scanner.laserscanner.FSSerial - DEBUG - M22;
2017-08-23 16:30:04,935 - fabscan.scanner.laserscanner.FSSerial - DEBUG - M05 R50 G50 B50;
2017-08-23 16:30:06,565 - fabscan.scanner.laserscanner.FSSerial - DEBUG - G04 T-44.4444444444 F900;
2017-08-23 16:30:06,572 - fabscan.scanner.laserscanner.FSCalibration - DEBUG - Calibration Position 51 of 74
```

mariolukas, 2017-08-23:
Which software Version are you using, and which version of the calibration sheet ?
Can you post a picture of your calibration sheet here ?

brummbaer, 2017-08-24:
Hope this is the right calibration sheet with 6x8 edges. The settings are in the config also.
[image]

mariolukas, 2017-08-25:
Yes this is the correct sheet. Can you have a look into your default.config.json. Check the number of columns and rows 6x8 there. Keep in mind to let the lid closed during the calibration.

brummbaer, 2017-08-31:
I have no FabScanPi HAT, so no security switch.  The switch is hardware based? Not implemented in firmware? 
`"calibration": {
        "weight_matrix": [],
        "dist_camera_matrix": [],
        "pattern": {
            "square_size": 11,
            "rows": 6,
            "columns": 8,
            "origin_distance": 35
        },
`

brummbaer, 2017-08-31:
Ok, i played around with the sheet, sometimes it is working. sometimes not. I can not figure it out. I think it is best to start with sheet a little bit more on the right. But what is the best position? The scanning result is always bad and not the same. Do i have to push the sheet up, as in your picture? http://mariolukas.github.io/FabScanPi-Server/images/Manual_2b.jpg

mariolukas, 2017-10-26:
Yes, try to adjust the cam. Use some cardboard or something else to tilt the cam to a center position.

closed 2018-09-23 by mariolukas