mariolukas/FabScanPi-Server #118: hardcoded steps/revolution in FSTurntable.py

doc2005, 2017-11-26:
in line 7 is the hardcoded value "STEPS_PER_ROTATION = 3200"
it seems that this prevents us from using different settings then 3200steps/revolution.
Settings in default.config.json were ignored for scan but not for calibration.

jwalt, 2017-12-30:
The magic constants of 577 and 1022 in FSCalibration are based on 3200 steps per revolution, too. This is a severe bug, as the Watterott kit comes with a 400 steps/revolution stepper motor, i.e. 6400 steps with default microstepping. As a result, the Watterott kit is currently plain broken.

Fixing the above constant and using something like "65/5*self.steps_five_degree" in place of those constants leads to working scans, finally.

awatterott, 2018-01-26:
The stepper motor from the FabScan Pi kit is a NEMA 17 with 200 steps/rev.

jwalt, 2018-01-27:
@awatterott There is definitely something wrong in your shop. I got a 17HM3410, your website claims it has 200 steps/rev, but the data sheet you offer as download says it's .9 deg/step, which is 400 steps/rev.

Based on my experience in getting the kit to work (it still produces inferior results, but at least the shape is now roughly correct), the data sheet is right. With 3200 steps (200*16 microsteps), output is utter garbage, objects come out twisted and incomplete. With 6400 it is at least roughly the right shape.

awatterott, 2018-01-27:
I will look into it. The 17HM3410 is listed in our ERP as 1.8° motor, but this is wrong.

jwalt, 2018-01-27:
For the record, I found the reason for my remaining quality problems, they are not related to this bug or the kit itself, it's a calibration problem.

As for the issue at hand, see PR #120

awatterott, 2018-01-28:
The 17HM3410 is included with the latest kits and we will update the software and documentation.
Thank you for the PR.

mariolukas, 2018-02-02:
I will replace the hardcoded values by config values in the new version. Thanks for reporting.

mariolukas, 2018-02-02:
@jwalt can you send me some pictures of the garbaged scans. I want to add them to the troubleshoot section of the documentation.

jwalt, 2018-04-02:
@mariolukas unfortunately (or luckily), calibration worked fine and I get decent scans. I don't have the distorted scans anymore, nor do I know what changed, another calibration just magically worked.

mariolukas, 2018-09-23:
fixed in v.0.5.0

closed 2018-09-23 by mariolukas