mariolukas/FabScanPi-Server #122: Power consumption Raspberry Pi 3 Model B+ and FabScanPI HAT

mariolukas, 2018-03-24:
## *Who* is the bug affecting?
Users of the new Raspberry Pi 3 Model B+

## *What* is affected by this bug?
Raspberry Pi 3 Model B+ has a higher power consumption, 
so that the FabScanPi HAT with stepdown will not serve 
enough power. 

## *When* does this occur?

## *Where* on the platform does it happen?
FabScanPi HAT.

## Expected behavior (i.e. solution)
People using the Raspberry Pi 3 Model B+ should not solder the stepdown and use
two different (12v supply for HAT and 5V Supply for Pi) port supplies.

thatSFguy, 2022-12-04:
What are the symptoms of not having enough power?  I'm having problems with hung scans, frozen PIcamera, etc.  so I was wondering if this could be a problem for me.  I'm using a DIY board, but probably very similar to your HAT implementation.  Thanks!

mariolukas, 2022-12-13:
The CPU slow down, sometimes freezes. There are different symptoms with under voltage. The pi is not able to work as expected.

closed 2024-04-08 by mariolukas