mariolukas/FabScanPi-Server #123: Feature Request - Atlas/Ciclop compatibility

wer-is-paul, 2018-04-15:
Hi,

I'm trying to implement this but I'm getting stucked on camera ready log. I can remote debug using pycharm. Anyone else working on this?

Cheers, 

Dominik

mariolukas, 2018-04-16:
Hey I am just working on supporting Atlas/Ciclop scanners. I already added USB Cam support in version 0.5.0 and i am making the firmware compatible right now.
Do you have a PyCharm Pro version? The CE Edition does not support remote debugging.

wer-is-paul, 2018-04-16:
Hi,

Yes, it's the pro version. But good to know. Is there a roadmap for the next version yet?

Cheers,

Dominik

mariolukas, 2018-04-19:
USB Cam Driver starts here 
https://github.com/mariolukas/FabScanPiServer/blob/master/src/fabscan/scanner/laserscanner/FSCamera.py#L288
It should work already. Just add USBCAM as camera type to the config. 
I will try to add a short remote debug tutorial to the docs this weekend.

mariolukas, 2019-07-03:
The ciclop support will be within the next release. But not working with the usb cam. You need to mount a raspberry pi cam on the ciclop Support for the atlas scanner will follow later, cause the atlas scanner uses a different electronics approach ( pi GPIO ). 
I added some pictures of the ciclop setup to the [forum](https://fabscan.org/community/development/second-laser-when/).

closed 2019-11-18 by mariolukas