mariolukas/FabScanPi-Server #124: Regular UI freezes, high cpu load, scans look like potato

ragflyer, 2018-04-26:
So we changed the connection to hostap mode due to connectivity issues (UI not opening in browser) and seem to have a stable SSH connection at least now.

After that, calibration ran fine once.

First scan hangs after a while, after 1 hour without UI update and 310% cpu load I reset the PI
Another scan ran fine but produced a potato from a simple toy
Another toy and it hangs again, reset again.
Another scan ran fine, again potato from 2nd toy.
As a third toy I tried a cuboid sponge, it looks like half sponge half potato.

Next morning, everything cooled down, start calibration, hangs at "2 minutes remaining" forever, the green calibration foot flickers green then yellow.
Browser refresh: cannot initialize UI.

Reset PI

Again calibration, hangs again at 2min, UI can be called but shows empty plate and "cancel scan" button, does not work, no video.
ssh login works, fabscanpi-server at 310%

last fabscanpi.log:
2018-04-26 06:30:19,509 - fabscan.scanner.laserscanner.FSCalibration - DEBUG - Calibration Position 28 of 74
2018-04-26 06:30:24,581 - fabscan.scanner.laserscanner.FSSerial - DEBUG - G04 T-44.4444444444 F900;
2018-04-26 06:30:24,584 - fabscan.scanner.laserscanner.FSCalibration - DEBUG - Calibration Position 29 of 74
2018-04-26 06:30:28,372 - fabscan.scanner.laserscanner.FSSerial - DEBUG - G04 T-44.4444444444 F900;
2018-04-26 06:30:28,375 - fabscan.scanner.laserscanner.FSCalibration - DEBUG - Calibration Position 30 of 74
2018-04-26 06:30:32,190 - fabscan.scanner.laserscanner.FSSerial - DEBUG - G04 T-44.4444444444 F900;
2018-04-26 06:30:32,192 - fabscan.scanner.laserscanner.FSCalibration - DEBUG - Calibration Position 31 of 74
2018-04-26 06:30:35,918 - fabscan.scanner.laserscanner.FSSerial - DEBUG - G04 T-44.4444444444 F900;
2018-04-26 06:30:35,920 - fabscan.scanner.laserscanner.FSCalibration - DEBUG - Calibration Position 32 of 74
2018-04-26 06:30:41,100 - fabscan.scanner.laserscanner.FSSerial - DEBUG - G04 T-44.4444444444 F900;
2018-04-26 06:30:41,102 - fabscan.scanner.laserscanner.FSCalibration - DEBUG - Calibration Position 33 of 74
2018-04-26 06:30:46,484 - fabscan.scanner.laserscanner.FSSerial - DEBUG - G04 T-44.4444444444 F900;
2018-04-26 06:30:46,485 - fabscan.scanner.laserscanner.FSCalibration - DEBUG - Calibration Position 34 of 74
2018-04-26 06:30:50,662 - fabscan.scanner.laserscanner.FSSerial - DEBUG - G04 T-44.4444444444 F900;
2018-04-26 06:30:50,664 - fabscan.scanner.laserscanner.FSCalibration - DEBUG - Calibration Position 35 of 74
2018-04-26 06:30:56,046 - fabscan.scanner.laserscanner.FSSerial - DEBUG - G04 T-44.4444444444 F900;
2018-04-26 06:30:56,048 - fabscan.scanner.laserscanner.FSCalibration - DEBUG - Calibration Position 36 of 74
2018-04-26 06:31:01,742 - fabscan.scanner.laserscanner.FSSerial - DEBUG - G04 T-44.4444444444 F900;
2018-04-26 06:31:01,745 - fabscan.scanner.laserscanner.FSCalibration - DEBUG - Calibration Position 37 of 74
2018-04-26 06:31:03,134 - fabscan.scanner.laserscanner.FSSerial - DEBUG - G04 T800 F900;
2018-04-26 06:33:45,185 - fabscan.server.websockets.FSWebSocket - DEBUG - Client disconnected
2018-04-26 06:33:45,604 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET / HTTP/1.1" 200 -
2018-04-26 06:33:45,616 - fabscan.server.FSHttpRequestHandler - DEBUG - MJPEG Stream closed by client.
2018-04-26 06:33:47,958 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET / HTTP/1.1" 200 -
2018-04-26 06:33:48,483 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET /style/lib.css HTTP/1.1" 200 -
2018-04-26 06:33:48,530 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET /style/app.css HTTP/1.1" 200 -
2018-04-26 06:33:48,542 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET /js/lib.js HTTP/1.1" 200 -
2018-04-26 06:33:48,698 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET /js/locales/en/i18n.js HTTP/1.1" 200 -
2018-04-26 06:33:48,724 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET /js/app.js HTTP/1.1" 200 -
2018-04-26 06:33:48,843 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET /icons/logo.png HTTP/1.1" 200 -
2018-04-26 06:33:48,926 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET /icons/icon_scan.svg HTTP/1.1" 200 -
2018-04-26 06:33:50,071 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET /icons/logo.png HTTP/1.1" 200 -
2018-04-26 06:33:50,173 - fabscan.server.websockets.FSWebSocket - DEBUG - New client connected
2018-04-26 06:33:50,181 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET /fonts/fontawesome-webfont.woff2?v=4.3.0 HTTP/1.1" 200 -
2018-04-26 06:33:50,518 - fabscan.server.FSHttpRequestHandler - DEBUG - 192.168.20.29 - "GET /api/v1/filters HTTP/1.1" 200 -
2018-04-26 06:33:50,731 - fabscan.FSScanner - DEBUG - Upgrade available: False 0.5.0
2018-04-26 06:33:50,901 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET /icons/icon_scan.svg HTTP/1.1" 200 -
2018-04-26 06:33:50,989 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET /icons/favicon.png HTTP/1.1" 200 -
2018-04-26 06:35:44,064 - fabscan.server.websockets.FSWebSocket - DEBUG - Client disconnected
2018-04-26 06:35:44,082 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET /icons/logo.png HTTP/1.1" 200 -
2018-04-26 06:35:46,436 - fabscan.server.websockets.FSWebSocket - DEBUG - New client connected
2018-04-26 06:35:46,912 - fabscan.FSScanner - DEBUG - Upgrade available: False 0.5.0
2018-04-26 06:35:46,953 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET /icons/icon_scan.svg HTTP/1.1" 200 -
2018-04-26 06:35:48,824 - fabscan.server.websockets.FSWebSocket - DEBUG - Client disconnected
2018-04-26 06:35:59,631 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET / HTTP/1.1" 200 -
2018-04-26 06:36:00,088 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET /style/lib.css HTTP/1.1" 200 -
2018-04-26 06:36:00,105 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET /style/app.css HTTP/1.1" 200 -
2018-04-26 06:36:00,123 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET /js/lib.js HTTP/1.1" 200 -
2018-04-26 06:36:00,191 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET /js/locales/en/i18n.js HTTP/1.1" 200 -
2018-04-26 06:36:00,197 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET /js/app.js HTTP/1.1" 200 -
2018-04-26 06:36:01,728 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET /icons/logo.png HTTP/1.1" 200 -
2018-04-26 06:36:01,756 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET /icons/icon_scan.svg HTTP/1.1" 200 -
2018-04-26 06:36:02,341 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET /icons/favicon.png HTTP/1.1" 200 -
2018-04-26 06:36:02,712 - fabscan.server.websockets.FSWebSocket - DEBUG - New client connected
2018-04-26 06:36:02,743 - fabscan.server.FSHttpRequestHandler - DEBUG - 127.0.0.1 - "GET /fonts/fontawesome-webfont.woff2?v=4.3.0 HTTP/1.1" 200 -
2018-04-26 06:36:03,130 - fabscan.server.FSHttpRequestHandler - DEBUG - 192.168.20.29 - "GET /api/v1/filters HTTP/1.1" 200 -
2018-04-26 06:36:03,190 - fabscan.FSScanner - DEBUG - Upgrade available: False 0.5.0
2018-04-26 06:37:35,449 - fabscan.server.websockets.FSWebSocket - DEBUG - Client disconnected

Laser is vertical, calibration sheet is one sided, glued flat on carton, 11mm boxes, camera works as shown, maybe a bit blurry (no protection on lens), table turns fine during calibration, seems well fixed to the motor shaft.

Any other ideas?

[image]
[image]
[image]
[image]
[image]
[image]

ragflyer, 2018-05-02:
Server: v.0.5.0 
Firmware: v.20170709

/etc/fabscanpi/default.config.json:
{
    "folders": {
        "www": "/usr/share/fabscanpi/", 
        "scans": "/home/pi/scans/"
    }, 
    "laser": {
        "numbers": 1
    }, 
    "scanner_type": "laserscanner", 
    "calibration": {
        "weight_matrix": [], 
        "dist_camera_matrix": [], 
        "pattern": {
            "square_size": 11, 
            "rows": 6, 
            "columns": 8, 
            "origin_distance": 35
        }, 
        "camera_matrix": [
            [
                1251.83, 
                0.0, 
                605.196
            ], 
            [
                0.0, 
                1258.142, 
                848.229
            ], 
            [
                0.0, 
                0.0, 
                1.0
            ]
        ], 
        "distortion_vector": [
            0.354, 
            -2.017, 
            0.004, 
            -0.004, 
            4.923
        ], 
        "laser_planes": [
            {
                "deviation": 0.021433795886643035, 
                "distance": 90.43523628037443, 
                "normal": [
                    0.8192565542352437, 
                    -0.04889458008552994, 
                    0.5713387947451624
                ]
            }
        ], 
        "platform_translation": [
            -0.17140579503327635, 
            49.614386467536896, 
            164.4640694418642
        ], 
        "platform_rotation": [
            [
                0.0, 
                0.9998354797193848, 
                -0.018138729126040996
            ], 
            [
                0.005011524963035171, 
                -0.018138501344051157, 
                -0.9998229240153163
            ], 
            [
                -0.9999874422299238, 
                -9.090269381288759e-05, 
                -0.0050107004655419425
            ]
        ]
    }, 
    "meshlab": {
        "path": "/usr/bin/"
    }, 
    "process_numbers": 4, 
    "turntable": {
        "steps": 3200, 
        "radius": 70
    }, 
    "camera": {
        "rotate": "True", 
        "type": "PICAM", 
        "vflip": "False", 
        "preview_resolution": {
            "width": 320, 
            "height": 240
        }, 
        "hflip": "True", 
        "resolution": {
            "width": 1640, 
            "height": 1232
        }
    }, 
    "serial": {
        "baudrate": 57600, 
        "autoflash": "True", 
        "port": "/dev/ttyAMA0"
    }, 
    "texture_illumination": 40
}
[image]

PavolxPetrovic, 2018-06-14:
I am getting this precise behavior as well. Now running the 0.5.0 server but it was the same on 0.4.3. Half of the scan seems to be fine and the other half almost "split and opened by 120 degrees", would be the best way to describe. Also the constant hang is not an uncommon issue. I am using raspberry pi 3 with the Hat and a Pi camera (1080p, 5MPx, 30fps).

s0cI, 2018-09-18:
Hello,

Has anyone solved this Problem yet? I'm also struggling a lot with the calibration and hanging of the server right now. I also use the Pi 3.

Best regards
Jan

PavolxPetrovic, 2019-05-25:
Hi, with the addition of the cross for camera in the new version i have seen that the camera is not aligned with the center of the turntable, when i have adjusted the camera to the center the issues mentioned above stopped. the camera is held by a 2 side adhesive strip that in my case has unstuck so the camera was angled to the side. now i am getting fine scans.

mariolukas, 2019-11-25:
fixed within v.0.8.0
- added more swap memory
- reduced number of used cpus in default config
- fixed some firmware bugs
- added temperature watch

reopen if needed.

closed 2019-11-25 by mariolukas