mariolukas/FabScanPi-Server #128: Critical, fatal error 'NoneType' object has to no attribute 'read'

PhistterbutInc, 2019-04-27:
## *Who* is the bug affecting?

User

## *What* is affected by this bug?

The server does not start

## *When* does this occur?

every time I reboot and try to start or restart the fabscanpi-server service

## *Where* on the platform does it happen?

I'm using the latest image as of 4/26/19 and a Rpi zero W.

## *How* do we replicate the issue?

Burn disk using win32 disk imager, insert SD into slot and power on the pi zero w. I've also tried to install the software manually as listed under the getting started page by adding the key and repo.

## Expected behavior (i.e. solution)

It should have worked considering I'm using the stock image provided. I didn't expect errors.

## Other Comments
Restarting the server service doesn't work either. Screen cap of errors included.
[image]

mariolukas, 2019-04-29:
Have you checked the serial port config in your default.config.json. The default value is

`/dev/ttyAMA0`

but in your case it should be 

`/dev/ttyACM0`

PhistterbutInc, 2019-05-02:
Yes, I've changed that value to ttyACM0

PhistterbutInc, 2019-05-15:
Do you need more info to help me or something??

mariolukas, 2019-11-25:
Could be a boot loader problem, if it still remains try to [flash the bootloader](https://fabscan.org/community/f-a-q/no-fabscanpi-hat-or-compatible-device-found/)

closed 2019-11-25 by mariolukas