mariolukas/FabScanPi-Server #129: Using HE3D Open Source Ciclop attached to Pi.  Is that possible?

mperino, 2019-07-31:
This is a Question, not a bug.
I have a HE3D Open Source Ciclop that is at best described as buggy using horus.  What Im wondering is could I use that as the basis of a FabScan?  It should be possible I think.. Attach it to a Pi and use the arduino module + stepper, one of the laser's, and the camera.

Thoughts?

PhistterbutInc, 2019-07-31:
I've tried a RAMPS board and so far it's a no go. Dev doesn't seem to care, look elsewhere.

mariolukas, 2019-08-01:
@mperino The next release (planed at the mid of august) will bring full ciclop support. All you need to to is add a pi and a pi camera. I am preparing the release since weeks now. You will be able to use the original Arduino (ZUM board).

> Dev doesn't seem to care, look elsewhere.

@PhistterbutInc since I am the only active developer on the project I am not able to realize every feature wish when it is coming up.  
But the new release brings also the opportunity to attach custom hardware boards. And this is a feature which was a long time in the dev pipeline. If you really want the RAMPS to be part of it please checkout Firmware and create a file called pins_RAMPS.h, define the pin outs of the RAMPS there and add it to the repository and do a pull request. It will be part of the next release. You can use the following file as a template.

https://github.com/mariolukas/FabScanPiFirmware/blob/master/FabScanArduinoFirmware/pins_FABSCANPI.h

The board can be defined in the default.config.json in the upcoming release. (see: https://github.com/mariolukas/FabScanPi-Server/blob/feature/ciclop/src/fabscan/config/default.config.json#L54)

By now the following boards will be supported. 

- FabScanPi HAT
- ciclop ZUM board
- Sanuinololu 
- CNC Shield

mperino, 2019-08-01:
@mariolukas that's great news!  Please consider me a tester and light dev (python mostly).

closed 2019-08-01 by mariolukas

PhistterbutInc, 2019-08-01:
@mariolukas sorry mate, I get sort of wasted sometimes....

I'm building from that list now and I'll see what I can put together to include RAMPS' pinout in your project. I did already start working on it but I want to get the server running first. I think building from this updated script will work out better because you added the lines for serial perms on ACM0. If it works for the RAMPS board I'll link you to the clean image file. You can do whatever you like with it. Thanks and sorry about that again.