mariolukas/FabScanPi-Server #130: Following clean install instructions leads to following error

PhistterbutInc, 2019-07-31:
## *Who* is the bug affecting?
user

## *What* is affected by this bug?
user

## *When* does this occur?
when attempting to install from source following instructions from: https://mariolukas.github.io/FabScanPi-Server/software/

## *Where* on the platform does it happen?
from stock Rasbian buster image installing from https://mariolukas.github.io/FabScanPi-Server/software/ instructions

## *How* do we replicate the issue?
install rasbian buster image and attempt to install based on instructions from:https://mariolukas.github.io/FabScanPi-Server/software/

## Expected behavior (i.e. solution)
I expected a clean install of the software: fabscanpi-server

## Other Comments
I received errors as follows:

Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python-opencv-tbb : Depends: libopencv-calib3d2.4-tbb but it is not going to be installed
                     Depends: libopencv-contrib2.4-tbb but it is not going to be installed
                     Depends: libopencv-features2d2.4-tbb but it is not going to be installed
                     Depends: libopencv-highgui2.4-tbb but it is not going to be installed
                     Depends: libopencv-legacy2.4-tbb but it is not going to be installed
                     Depends: libopencv-objdetect2.4-tbb but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

mariolukas, 2019-08-01:
That is because the FabscanPi Software depends on a OpenCV version with tbb support. TBB is needed for multicore image processing. The Raspbian default packages do not support multicore. 
I added the packages to the fabscanpi source repository archive.fabscan.org. 
You need to add it to your apt configuration

`deb http://archive.fabscan.org/ <testing or stable here> main`

It you really want to start with a clean installation you should try the FabScanPi image build script. It based on the official raspbian build script and adds an extra stage for the fabscan stuff. You can find it here https://github.com/mariolukas/FabScanPi-Build-Raspbian

There are some other special configurations which need to be done to get the software running. Have a look at the stage folder if you want to configure it by yourself. Otherwise try to build the image with the build script.

PhistterbutInc, 2019-08-01:
Alright, I'll give that a shot. I had given up on trying to get my previous error worked out and thought I'd try this method. Now I'll try that method. Thanks.

mariolukas, 2019-08-01:
There is nothing special to do. Just change the port in the FabScanPi Config file from /dev/ttyAMA0 to something like /dev/ttyUSB0

PhistterbutInc, 2019-08-05:
The error seems to be an issue with pi-gen. I've tried for three days to make the image from their scripts using the docker build script and still nothing. The error still revolves around the /pi-gen/work/ folder not being created even I believe. Oh well, I opened an issue and hopefully it gets resolved soon.

mariolukas, 2019-08-05:
This issue is more related to https://github.com/mariolukas/FabScanPi-Build-Raspbian/issues/15
so I think we should discuss it on the build script repo.

closed 2019-08-05 by mariolukas