mariolukas/FabScanPi-Server #139: The FabScanPi server is currently not running

KayIO, 2019-12-27:
## *Who* is the bug affecting?

## *What* is affected by this bug?
Fabscan Server not starting up... Always "The FabScanPi server is currently not running" is shown.

## *When* does this occur?
Everytime

## *Where* on the platform does it happen?
Raspberry 3 with FabscanPi Image

## *How* do we replicate the issue?
Just install with clean image

## Expected behavior (i.e. solution)
Webserver should running or give at least an error message.
The basic configuration is:
Raspi 3 with latest FabscanPi Image
Connected and working Raspi Cam V1
Ciclop Scanner with Arduino Board (flashed FW from fabscanpi already manually)
Used configuration for ciclop. Already changed the desired baudrate for the USB connection a couple of times and it seems to be 115200 to be the right one, as ther eis no error.
After start it needs a couple of seconds to start also the FabscanPi Server, but the web interface shows only "The FabScanPi server is currently not running"
But when I check with the command 
ps -ef | grep -i fabscanpi | grep -i python
I get this
<-----
/usr/bin/python /usr/bin/fabscanpi-server --config=/etc/fabscanpi/default.config.json --settings=/etc/fabscanpi/default.settings.json --logfile=/var/log/fabscanpi/fabscanpi.log
----->
So it seems to be running...
The Log File says:
<-----
2019-12-26 23:56:58,669 - fabscan.server.FSScanServer - INFO - FabScanPi-Server 0.8.0+201911082051
2019-12-26 23:56:58,671 - fabscan.scanner.interfaces - DEBUG - Scanner Type is: laserscanner
2019-12-26 23:56:59,351 - fabscan.scanner.laserscanner.driver.FSSerial - DEBUG - Port in Config found
2019-12-26 23:56:59,354 - fabscan.scanner.laserscanner.driver.FSSerial - DEBUG - Latest available firmware version is: v.20190811
2019-12-26 23:57:02,752 - fabscan.scanner.laserscanner.driver.FSSerial - DEBUG - Trying to connect Arduino on port: /dev/ttyUSB0
----->

When I start the server manually by command line the following information is shown
<-----
avrdude: AVR device initialized and ready to accept instructions
Reading | ################################################## | 100% 0.00s
avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: safemode: Fuses OK (E:00, H:00, L:00)
avrdude done.  Thank you.
----->

I've done alle necessary checks I know.
Raspi cam is running... USB connection is working ... there are no errors in the log file.
I have no clue, what is not working...

mariolukas, 2019-12-27:
This is a well known bug in v.0.8.0. The server is not flashing the firmware and gets stuck in a deadlock while trying connecting to the HAT. This only happens when the HAT is in an initial state. No firmware installed with a plain bootloader on it. A workaround might be to flash the latest firmware manually. 

```
sudo /ect/init.d/fabscanpi-server stop

sudo avrdude-autoreset -D -V -U flash:w:/usr/lib/python2.7/dist-packages/fabscan/firmware/fabscanpi_v.20190811.hex:i -b 115200 -carduino -pm328p -P/dev/ttyAMA0

sudo /ect/init.d/fabscanpi-server start 
```

KayIO, 2019-12-27:
I already tried that. Yes, I use the current 0.8.0 and I used the ciclop firmware for the ciclop board:
`sudo avrdude-autoreset -D -V -U flash:w:/usr/lib/python2.7/dist-packages/fabscan/firmware/ciclop_v.20190811.hex:i -b 115200 -carduino -pm328p -P/dev/ttyUSB0`
This results in succesfull flashing the firmware with no errors.

But the output is always stuck with 
`pi@fabscanpi:~ $ /etc/init.d/fabscanpi-server start`
results in 
`avrdude-original: Using autoreset DTR on GPIO Pin 7`
`fabscanpi-server started`
But the web interface still says 
`The FabScanPi server is currently not running`

Even when I set 
`"autoflash": "False",`
in the config file. There is no change.

Is there any other work around? Or at least a workaround to see more details in any log file ? Setting another debug level or similar ?

mariolukas, 2019-12-27:
Sure next you can try to patch the serial file until the bugfix version is ready to use. 
Change to: 
` /usr/lib/python2.7/dist-packages/fabscan/scanner/laserscanner/driver/`
and edit the file `FSSerial.py `

And change this line: 

https://github.com/mariolukas/FabScanPi-Server/blob/develop/src/fabscan/scanner/laserscanner/driver/FSSerial.py#L66

to

self._serial = serial.Serial(str(self._port), int(self._baudrate), timeout=3)

The pyserial [docs](https://pyserial.readthedocs.io/en/latest/pyserial_api.html) say 

> timeout = None: wait forever / until requested number of bytes are received
> timeout = 0: non-blocking mode, return immediately in any case, returning zero or more, up to the requested number of bytes
> timeout = x: set timeout to x seconds (float allowed) returns immediately when the requested number of bytes are available, otherwise wait until the timeout expires and return all bytes that were received until then.

default it None, and I think this is the problem. In previous versions there was a timeout in this line, but I accidentally removed it while refactoring, and because the firmware was already running on my setup it worked for my scanner.

KayIO, 2019-12-27:
I patched the file and found, that I also had a problem obviously with haproxy which did not connect to the fabscanpi-server. 
I corrected it (it was a typo by me) and it's working now. 
I think the patch also seems to help to prevent the "loop of death". I set the timeout to 3 as recommended.
Thank you very much

mariolukas, 2019-12-27:
> I corrected it (it was a typo by me) and it's working now.
If you are running in trouble with the ciclop, please open a new thread in the forum. We can get a kind of documentation. I already know that you will have a next question ;-)

closed 2019-12-27 by mariolukas