mariolukas/FabScanPi-Server #140: Camera rotation does not work and results in stuck server

KayIO, 2019-12-28:
As you already promised there will be more with ciclop ;)

I have a new issue when try to deactivate the camera rotation. When I change 
`"rotate": "False",`
in the config file, the server starts but is stuck at this point. It just activate one laser and doesn't respond anymore to actions in webgui.

The Logfile says nothing special...
_2019-12-28 23:13:47,990 - fabscan.server.FSScanServer - INFO - FabScanPi-Server 0.8.0+201911082051
2019-12-28 23:13:47,992 - fabscan.scanner.interfaces - DEBUG - Scanner Type is: laserscanner
2019-12-28 23:13:48,582 - fabscan.scanner.laserscanner.driver.FSSerial - DEBUG - Port in Config found
2019-12-28 23:13:48,589 - fabscan.scanner.laserscanner.driver.FSSerial - DEBUG - Latest available firmware version is: v.20190811
2019-12-28 23:13:51,816 - fabscan.scanner.laserscanner.driver.FSSerial - DEBUG - Trying to connect Arduino on port: /dev/ttyUSB0
2019-12-28 23:13:54,846 - fabscan.scanner.laserscanner.driver.FSSerial - DEBUG - Installed firmware version: v.20190811
2019-12-28 23:13:54,848 - fabscan.scanner.laserscanner.driver.FSSerial - INFO - FabScanPi is connected to FabScanPi HAT or compatible on port: /dev/ttyUSB0
2019-12-28 23:13:56,872 - fabscan.scanner.laserscanner.driver.FSSerial - DEBUG - Connection baudrate is: 14400
2019-12-28 23:13:56,873 - fabscan.scanner.laserscanner.driver.FSSerial - DEBUG - Firmware flashing baudrate is: 115200
2019-12-28 23:13:56,874 - fabscan.scanner.laserscanner.FSHardwareController - DEBUG - Reset FabScanPi HAT...
2019-12-28 23:13:57,176 - fabscan.scanner.laserscanner.driver.FSSerial - DEBUG - M18;
2019-12-28 23:13:57,479 - fabscan.scanner.laserscanner.driver.FSSerial - DEBUG - G07;
2019-12-28 23:13:57,980 - fabscan.scanner.laserscanner.driver.FSCamera - DEBUG - Cam Stream with Resolution (640, 480) stopped
2019-12-28 23:13:57,982 - fabscan.scanner.laserscanner.FSHardwareController - DEBUG - Reset FabScanPi HAT...
2019-12-28 23:13:58,284 - fabscan.scanner.laserscanner.driver.FSSerial - DEBUG - M18;
2019-12-28 23:13:58,586 - fabscan.scanner.laserscanner.driver.FSSerial - DEBUG - G07;
2019-12-28 23:13:58,587 - fabscan.scanner.laserscanner.FSHardwareController - DEBUG - Hardware controller initialized...
2019-12-28 23:13:58,781 - fabscan.scanner.laserscanner.FSScanProcessor - INFO - Laser Scan Processor initilized...FSScanProcessor (urn:uuid:b973c991-03df-4e6f-b929-84192bc3b4e3)
2019-12-28 23:13:58,782 - fabscan.server.FSWebServer - DEBUG - /usr/share/fabscanpi/
2019-12-28 23:13:58,805 - fabscan.FSScanner - INFO - Job scheduler started.
2019-12-28 23:13:58,806 - fabscan.FSScanner - INFO - Scanner initialized...
2019-12-28 23:13:58,806 - fabscan.FSScanner - INFO - Number of cpu cores: 4
2019-12-28 23:13:58,807 - fabscan.lib.util.FSDiscovery - INFO - Registering ip 192.168.178.91 to discovery service.
2019-12-28 23:13:59,306 - fabscan.server.services.websocket.FSWebSocketHandler - DEBUG - New client connected
2019-12-28 23:13:59,849 - fabscan.lib.util.FSUpdate - ERROR - <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:727)>
2019-12-28 23:13:59,852 - fabscan.FSScanner - DEBUG - Upgrade available: False 0.8.0+201911082051
2019-12-28 23:13:59,860 - fabscan.scanner.laserscanner.FSScanProcessor - DEBUG - Checking Hardware connections
2019-12-28 23:14:01,515 - fabscan.lib.util.FSDiscovery - INFO - Successfully registered to find.fabscan.org
2019-12-28 23:14:01,650 - fabscan.FSScanner - INFO - Added discovery scheduling job.
2019-12-28 23:14:16,439 - fabscan.scanner.laserscanner.driver.FSSerial - DEBUG - M05 R0 G0 B0;
2019-12-28 23:14:16,534 - fabscan.scanner.laserscanner.driver.FSCamera - DEBUG - Cam Stream with Resolution (640, 480) started
2019-12-28 23:14:16,837 - fabscan.scanner.laserscanner.driver.FSSerial - DEBUG - M21;_

When I change it back to 
"rotate": "True",
it works again but rotated. 

The problem with the ciclop is, that it's better to lead the cam cable up downwards, than left or right, as there is no real nice space for it.

I use the 5MP RaspiCam Version.
I also tried to set the cam resolution, but it doesn't change anything. I also tried the change width and height, but this leads to a ressource error.

BTW: I changed also the flashing port in the "FSSerial.py" from ttyAMA0 to ttyUSB0. This seems to solve the flashing error for the cheap ciclop board in the issue #139

mariolukas, 2019-12-29:
I will have a closer look to the camera config params. I think it would be better to switch to the forum for further ciclop queststions. 
https://fabscan.org/community/case/ciclop/

mariolukas, 2020-06-23:
solved within the upcoming version 0.9.0 see dev branch

closed 2020-06-23 by mariolukas