mariolukas/FabScanPi-Server #145: The FabScanPi server is currently not running

Faraznassajian, 2021-01-30:
Hi
I am very new to this device. 
I have already two 3d printers and raspberry pi connected to them and use octoprint.
I decided to buy a 3d scanner which is affordable for me and many people suggested fabscanpi.
I bought a kit and soldered and assembeled as on website.
I updated everything by the website suggestion.
I installed the server deb file as well.
Everything seems to work fine but I stuck here with this message.
These commands are performed:
**sudo /etc/init.d/fabscanpi-server start**
result : **fabscanpi-server: running.**
in the log file I get these errors:
**ERROR - fabscan.scanner.laserscanner.driver.FSSerial : communication error on port /dev/ttzAMA0 try other flashing baudrate than 115200. Maybe corrupted bootloader.
ERROR - fabscan.scanner.laserscanner.driver.FSSerial : Fatal fabscanpi HAT or compatible connection error : 'NoneType' object has no attribute 'isOpen'**
So I thought maybe the problem is with the baudrate and I edited the json file and changed it to 57600 once and to 57200 another time but no change.
what is the problem?
sorry I am noob in this matter

mariolukas, 2021-01-30:
Sorry but the documentation for installing with the deb packages is outdated. I will try to update the manual installation (deb package way) instructions in the upcoming days.

Faraznassajian, 2021-01-30:
You think the problem is with the Deb package?
I installed the latest image file from website as well I performed an update as well.
Before installing Deb file the problem was there...

mariolukas, 2021-01-30:
Are you sure that you are using the correct serial port? Have a look at 

>ERROR - fabscan.scanner.laserscanner.driver.FSSerial : communication error on port /dev/ttzAMA0 try other flashing 
 
/dev/ttzAMA0 should be /dev/ttyAMA0

Faraznassajian, 2021-01-30:
Yes sir, actually my keyboards bad. But I checked it first without any changes or upgrades but no result. I decided to order another HAT and solder it again but more carefully. I think one of the pins soldering might have some problems. I spent 45 hours on it solve or find the problem but no success so the only thing should be my mistake in soldering. I will keep this thread updated as soon as I get the new HAT and solder it. 
Thanks again for your response.

mariolukas, 2021-02-04:
Have you fixed this ?

Faraznassajian, 2021-02-05:
Yes sir, I got the new HAT and soldered it and the problem is solved but now after calibration, I tried to scan my USB charger as first scan and this odd thing happened 
Link:
https://drive.google.com/file/d/1p0uTtjeh2SE5KsPcOk93FkTzs7HbQ06j/view?usp=drivesdk
I followed the documentations but no success 
I calibrated the device 10 times and no success 
I scanned with different resolution and thresholds but still no success...
What shall I do sir?
Thank you for your help

mariolukas, 2021-02-07:
Moved to the community Forum

closed 2021-02-07 by mariolukas