mariolukas/FabScanPi-Server #146: Calibration

Faraznassajian, 2021-02-03:
I calibrated the device as mentioned on the website at the end it says calibration finished but as soon as I click on scan to start scan something it says scan stopped!
Not calibrated and follow the website...
I tried it over 4 hours and no success 
I used two different SD cards 
I used a windows based and a Mac to flash SD card 
The laser is working perfectly 
The camera is focused and nice 
The ring light is charming 
The turntable is working perfect 
There is no error in fabscanpi.log
It does not scan it says not calibrated...

mariolukas, 2021-02-04:
How many lasers do you use, and how many lasers are configured as number of lasers in the config file? If you are using one laser you should change the number of lasers to 1 in the config, default is 2. Otherwise the scanner will end up in a faulty calibration.

Faraznassajian, 2021-02-04:
Thank you sir.
It worked 
You are right and I should follow the forum the problem is, I am a medical doctor and I do this as a hobby after work when I get home and I am so tired to look for it a lot and follow everything.
Thank you for your precious time and great help and support.

closed 2021-02-05 by mariolukas